ebroecker/canmatrix #708: New Release?

pierreluctg, 2023-06-20:
Hi @ebroecker, when do you plan the next canmatrix release?

ebroecker, 2023-06-22:
Hi @pierreluctg,

I plan it the next weeks maybe days.
I want to fix/merge some small issues and afterwards create a new release (probably the last one with some py2 support).

Do you have some special wishes for the next release?

pierreluctg, 2023-06-22:
@ebroecker my focus is on ARXML parsing, so I am looking at a release that includes at least https://github.com/ebroecker/canmatrix/pull/692

Thank you

ebroecker, 2023-07-18:
OK, I am near to create the next release.
Maybe even today

ebroecker, 2023-07-19:
Done

closed 2023-07-19 by ebroecker

pierreluctg, 2023-07-24:
Thank you @ebroecker

elupus, 2023-11-07:
Nice to see a new release. Something however went wrong on github. The 1.0 named release, points to the 0.95 tag. I assume it ended up being renamed instead of creating a new release for the new 1.0 tag.

ebroecker, 2023-11-08:
> Hi @ebroecker, when do you plan the next canmatrix release?

Hi…

Really? I cannot find this problem in the repo?

elupus, 2023-11-08:
Have a look at https://github.com/ebroecker/canmatrix/releases/tag/0.9.5

Its the latest release on GitHub