ebroecker/canmatrix #711: Hello, why do I get a multiplexed signal to a normal signal when converting between ARXML and DBC files

wpphh, 2023-06-26:


ebroecker, 2023-06-26:
Hi @wpphh 

could you please provide additional information?

Did you convert 
ARXML  -> dbc
or  dbc -> ARXML -> dbc

or something else?

Do you have example files?

wpphh, 2023-06-27:
dbc -> ARXML -> dbc

wpphh, 2023-06-27:
ARXML -> dbc.Both of these will

ebroecker, 2023-06-27:
Hi @wpphh 

dbc -> ARXML
exporting ARXML: This implementation is _very basic_  (https://canmatrix.readthedocs.io/en/latest/formats.html#export) and has no support for multiplexed signals.

ARXML -> dbc
If you have a correct ARXML the import (and dbc export) of multiplexed signals should work.
Do you have an example, or can you cook down the missbehaviour?
Thanks in advance

wpphh, 2023-06-28:
Arxml ->dbc,
This is what I converted using camatrix：
BO_ 300 BMS_PTCANFD_20ms_ConFrP01: 64 ESS
 SG_ Header_ID : 7|24@0- (1,0) [-8388608|8388607] "" Vector__XXX
 SG_ Header_DLC : 31|8@0- (1,0) [-128|127] "" Vector__XXX
 SG_ IBMSPackHVInfoChecksum m65797 : 39|8@0+ (1,0) [0|255] "null" Vector__XXX
 SG_ IBMSPackHVInfoCounter m65797 : 43|4@0+ (1,0) [0|15] "null" Vector__XXX

Then I compared it with other tools

SG_ Header_ID M : 7|24@0- (1,0) [-8388608|8388607] "" Vector__XXX
SG_ Header_DLC : 31|8@0- (1,0) [-128|127] "" Vector__XXX
 SG_ IBMSPackHVInfoChecksum m65797 : 39|8@0+ (1,0) [0|255] "null" Vector__XXX
 SG_ IBMSPackHVInfoCounter m65797 : 43|4@0+ (1,0) [0|15] "null" Vector__XXX

ebroecker, 2023-06-28:
Hi @wpphh,

thanks for these details, both versions look pretty equal, in the first one "M" for Multiplexer is missing.
I just changed something in the dbc-export - so I'll double check if I did something worse with this change...

By the way, this seems to be a CONTAINER-IPDU thing in ARXML which canmatrix tries to convert in something working.
Is "other tools" an other version of canmatrix or a really other tool?

ebroecker, 2023-06-28:
Hi @wpphh 

could you try  branch issue_711?

pip install git+https://github.com/ebroecker/canmatrix.git@issue_711

Thanks

wpphh, 2023-06-30:
arxml->dbc reports an error
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2103' in position 48: ordinal not in range(256)

ebroecker, 2023-06-30:
Hi @wpphh,

thanks for testing.
This seems to be another error. Some Unicode character cannot coded to latin-1 (seems to be °).
Maybe you have some more context for this error (some source line?)

could you try converting with this additional command line option: `--ignoreEncodingErrors`

thanks

wpphh, 2023-07-03:
Hello, the problem has been resolved and the "M" of the multiplexer is also available. However, after converting with the canconvert 1. arxml 2. dbc -- ignoreEncodingErrors command, some other prompts appear:

WARNING - dbc - Signal ICC_ PTCANFD_ 50ms_ SigGW_ FrP01:: IBMSOnbdChrgrAltCrntLmtBkup_ PTCANFD name exceeds 32 characters, consistent updating the signal name within the character limit (Max 32 characters)
WARNING - dbc - Signal BMS_ PTCANFD_ 50ms_ ConFrP01:: IBMS_ 50ms_ VHILSts_ Group01_ Reserve01 name exceptions 32 characters, consistent updating the signal name within the character limit (Max 32 characters)
WARNING - dbc - Signal BMS_ PTCANFD_ 50ms_ ConFrP01:: IBMS_ 50ms_ VHILSts_ Group01_ Reserve02 name exceptions 32 characters, consistent updating the signal name within the character limit (Max 32 characters)
WARNING - dbc - Signal BMS_ PTCANFD_ 50ms_ ConFrP01:: IBMS_ 50ms_ VHILSts_ Group01_ Reserve03 name exceptions 32 characters, consistent updating the signal name within the character limit (Max 32 characters)
WARNING - dbc - Signal ESS_ PTEXTDCAN_ 50ms_ FrP23:: IBMSOnbdChrgrAltCrntLmtBkup_ PTEXTDCAN name exceptions 32 characters, consistent updating the signal name within the character limit (Max 32 characters)
INFO - convert - done

Do these have an impact on the results?

ebroecker, 2023-07-03:
Hi @wpphh ,

thanks for testing again.

These warnings are OK.
The names in ARXML for the signals are longer than allowed in dbc, this is why they are shortended.
The long name is stored in dbc also (in Attribute SystemSignalLongSymbol).

P.S: Was there some autocorrect in your post? The Warning-Message should say "..name exceeds..." not "... name exceptions ..."

wpphh, 2023-07-07:
hi ,arxml->dbc
The arxml file directory is so deep that it will report an error：
C:\Users\wph>canconvert d:\arxml\1\222\3.arxml d:\arxml\1\222\3.dbc
ldf is not supported
xls is not supported
xlsx is not supported
yaml is not supported
INFO - convert - Importing d:\arxml\1\222\3.arxml ...
Traceback (most recent call last):
  File "c:\users\wph\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\users\wph\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\wph\AppData\Local\Programs\Python\Python39\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\cli\convert.py", line 155, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\arxml.py", line 2005, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\arxml.py", line 1899, in decode_can_helper
    frame = get_frame(frameTrig, ea, multiplex_translation, float_factory, headers_are_littleendian)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\arxml.py", line 1635, in get_frame
    get_signals(pdu_sig_mapping, new_frame, ea, None, float_factory, signal_triggerings=isignaltriggerings_of_current_cluster)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\arxml.py", line 1237, in get_signals
    (values, factor, offset, unit_elem, const) = decode_compu_method(compu_method, ea, float_factory)
  File "c:\users\wph\appdata\local\programs\python\python39\lib\site-packages\canmatrix\formats\arxml.py", line 1034, in decode_compu_method
    if len(desc) == 0:
TypeError: object of type 'NoneType' has no len()

canconvert d:\arxml\3.arxml d:\arxml\1\222\3.dbc，This way no error is reported

wpphh, 2023-07-09:
I don't know how this mistake was caused

ebroecker, 2023-07-18:
OK, I'll close this issue.

For the problem:
`if len(desc) == 0:
TypeError: object of type 'NoneType' has no len()`
I have a fix already, will provide it with pr: #717

closed 2023-07-18 by ebroecker