ebroecker/canmatrix #714: Does canmatrix support parsing of Ethernet ARXML files?

mominggame, 2023-07-08:
After loading an Ethernet ARXML file using 'dbs=canmatrix.formats.loadp(file_path)', the length of 'dbs' is 0.

ebroecker, 2023-07-09:
Hi @mominggame ,

please try:

`cluster = canmatrix.formats.arxml.load(arxml, decode_ethernet=True)` 

See also:
https://github.com/ebroecker/canmatrix/issues/443#issuecomment-1619973352

mominggame, 2023-07-10:
I was able to parse ETH using your method, but I noticed that there was no IP address in the results. Could you please help me understand why this is the case? Thank you.

ebroecker, 2023-07-10:
Hi @mominggame ,

sure : the ethernet-support is **very** basic. IP-addresses are not extracted (yet).

Maybe you describe your use-case a bit more in detail, than an implementation would be possible.

mominggame, 2023-07-12:
Thank you very much for your answer. Can the command 'cluster = canmatrix.formats.arxml.load(arxml, decode_ethernet=True)' be used to convert the cluster to JSON format? Is there any documentation available to help with this?"

ebroecker, 2023-07-12:
Hi @mominggame 

Yes, is possible:
`convert.py --arxmlEthernet some_arxml_file.arxml some_json_file.json`

But again, probably onle very limited information are extracted from ARXML

see also:
https://github.com/ebroecker/canmatrix/issues/443#issuecomment-1619920327

closed 2023-07-12 by mominggame