ebroecker/canmatrix #718: J1939 Digital Annex to DBC

mattie47, 2023-07-20:
Hi there,
I just wanted to clarify whether the intentions of the XLS/XLSX functionality for conversion is supposed to work with the J1939 Digital Annex files?

e.g. J1939DA_202305.zip / J1939DA MAY23.xlsx

We recently purchased the latest copy of the DA, however having tried this tool, and https://github.com/nmfta-repo/pretty_j1939 for conversion to a DBC, I haven't been having much luck. Pretty_j1939 should do it, but hasn't been updated to work with later versions of the spreadsheet which seem to break some compatibility (even if you get past the lack of xlsx support) 

So before attempting any further on canmatrix, I just wanted to confirm whether the DA files should be able to be converted to a DBC with canmatrix?

Thanks,

Matt

ebroecker, 2023-07-20:
Hi @mattie47 ,

I am quite sure, canmatrix cannot do the conversion.
I never ha a DA in my hands, so I'd wounder if this works...

But maybe it's not to hard to implement, if I could get some basic information about this xlsx.
things like column-headers would help a lot

mattie47, 2023-07-24:
Thanks Eric.

Slide 76, and slide 77 have a couple screenshots of the historic xls 2020 version:
https://www.cybertruckchallenge.org/wp-content/uploads/2022/06/Introduction-to-SAE-J1939-CyberTruck-2022.pdf 

I'm not too sure how much can be shared, but I will see if I can provide a sanitised file with just the headers etc. 

Worthy of note for https://github.com/nmfta-repo/pretty_j1939 is the author had always intended on trying to make it work with canmatrix (see last section of the readme).

https://github.com/andrewdodd/decoda is another tool which also built on pretty_j1939:
_However, if you own a copy of the SAE Digital Annex ([link](https://www.sae.org/standards/content/j1939da_202201/)), then this library provides a number of bundled [scripts](https://github.com/andrewdodd/decoda/tree/main/src/decoda/sae_spec_converter) to convert the XLS file to a suitable JSON spec file. T**he code for these scripts borrows heavliy from the [pretty_j1939]**(https://github.com/nmfta-repo/pretty_j1939) library, but it has been adjusted to work in a slightly different way._

MatinF, 2023-10-19:
We convert the J1939 Digital Annex to DBC format for every new revision, in a legal collaboration with SAE. You can purchase the DBC file here: https://www.csselectronics.com/products/j1939-dbc-file

In my opinion canmatrix is not well suited to handle this as the J1939 DA is filled with various syntax inconsistencies and issues. It's a lot of manual work to ensure the output is right.

mattie47, 2023-12-20:
Yeah it's quite a pain unfortunately. I wish I'd reached out here first to confirm what exact support there currently was, rather than just buying the spreadsheet. Things seem to be a lot simpler when they're in a DBC file for consumption in other tools, rather than the XLS/XLXS directly from SAE.

We had some shifting priorities internally so I haven't had a look at this much further.

A colleague did make a couple minor changes to pretty_j1939 to make it work with the latest DA though, however this was a pretty minimal change, and still requires converting the XLXS to XLS first:

https://github.com/SherifRiad/pretty_j1939

closed 2024-07-18 by ebroecker