ebroecker/canmatrix #721: Ethernet arxml parser

ApoorvaSatish, 2023-07-25:
Hello,

Have you developed ethernet arxml parser?

ebroecker, 2023-07-25:
Hi @ApoorvaSatish ,

depends what you need.

#283 :

` convert.py --arxmlEthernet some_arxml_file.arxml some_excel_file.xlsx`

ApoorvaSatish, 2023-07-25:
Hi,

Is the above arxml parser developed for dynamic multiple PDU handling?
I was looking for SOME/IP related arxml parser.

ebroecker, 2023-07-25:
Hi @ApoorvaSatish,

No idea what exactly you need, but maybe ` --arxmlIgnoreClusterInfo` can help you:

`convert  --arxmlIgnoreClusterInfo some.arxml some.xlsx` <= or whatever format you need.

ApoorvaSatish, 2023-08-07:
Hi,
Do you have a sample ethernet arxml which you can share?

ebroecker, 2024-07-18:
No sorry - hard to get

closed 2024-07-18 by ebroecker

saravanan-1988, 2025-04-17:
script executed but excel not generated, can you please help here

[image]