ebroecker/canmatrix #725: Question regarding frame.arbitration_id and MF4 file message ID'S

MatheusFreitas25, 2023-08-12:
I'm trying to match the id of the messages I have inside my CAN Channel with the id of the messages I have inside the DBC file.

however when I read the Arbitration ID inside the DBC I get only integer values ​​with 9 digits(But when I open the DBC as txt all of them have 10), for example:

ArbitrationId(id=418381865, extended=True)

and when I read the ids inside the MF4 files I get message ids with 10 digits example:

2564423920

How do I make this match?

ebroecker, 2023-09-25:
Hi @MatheusFreitas25,

I think you have some mismatsches in the IDs you provide.
But I think what you look for is the 'compound' id.

This 'compound' id is used if you want to store the arbitration ID in one integer, and also store the information 'extended' in the same integer.

It is usually done like this:
```python
mask = 1 << 31
id = 418381865  # id without exended information
print(id | mask)
```
but than Id 418381865 should be 2565865513 (with coded extended bit)

Even better in hex:
Id 0x18f00029,
compound: 0x98f00029

and the other ID: 2564423920 
is 2564423920 (extended, compound bit removed)

in hex: 0x98da00f0
without compound-bit: 0x18da00f0

this compound information seems to be useless, if you have such high arbitration IDs because they are implicite clear to be extended.
But keep in Mind, there are for example two arbitration IDs with the ID 1.
ID 1, extended = false  (0x1)
and
ID 1, extended = true (0x80000001)

Hope thats the info you need ...

MatheusFreitas25, 2024-04-08:
Thank you very much! This information will certainly be very useful.

closed 2024-04-08 by MatheusFreitas25