ebroecker/canmatrix #726: IndexError: list index out of range canconvert from .dbc to xlsx

TheBossAmine, 2023-10-04:
File "C:\Users\Axxxx\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\canmatrix.py", line 2317, in enum_attribs_to_values
    frame.attributes[define] = self.frame_defines[define].values[int(frame.attributes[define])]
IndexError: list index out of range

python version 3.10.10

Hello, I am trying to convert my dbc file into an xlsx format, the code seems to be crashing each time.

Thank you.

ebroecker, 2023-10-27:
Hi @TheBossAmine 

Some value which is set in one frame is not in the corresponding value-table (ENUM).
I think - but I'm not sure - this could be an error in your dbc.  Is candb++ fine with the dbc?

You are not able to share the dbc (maybe as pn?).

If not, maybe I'll create a special-debug-version for you, to find out the root cause...

ebroecker, 2024-06-19:
no news here ... - may be fixed already... - if not - please reopen!

closed 2024-06-19 by ebroecker