ebroecker/canmatrix #729: DBC issue: Unable to mix J1939 PGNs with standard CAN IDs

MatinF, 2023-10-19:
We use canmatrix in various ways, including through asammdf. 

When we try to convert attached log file with attached DBC, we get an issue because canmatrix appears to be unable to handle the fact that the DBC file contains both a standard CAN ID and a J1939PG style CAN ID. This is not an issue in e.g. CANDB++ or Kvaser's Database Editor, but provides below error when we try to decode in the asammdf GUI:

```
Traceback (most recent call last):
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\asammdf\gui\utils.py", line 251, in run
    result = self.function(*self.args, **self.kwargs)
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\asammdf\gui\widgets\file.py", line 1858, in extract_bus_logging_thread
    result = self.mdf.extract_bus_logging(
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\asammdf\mdf.py", line 4939, in extract_bus_logging
    out = self._extract_can_logging(
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\asammdf\mdf.py", line 5016, in _extract_can_logging
    messages = {
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\asammdf\mdf.py", line 5018, in <dictcomp>
    message.arbitration_id.pgn,
  File "C:\Users\marti\AppData\Local\Programs\Python\Python39\lib\site-packages\canmatrix\canmatrix.py", line 634, in pgn
    raise J1939needsExtendedIdetifier
canmatrix.canmatrix.J1939needsExtendedIdetifier
```

Here are the files:
[obd2-test.zip](https://github.com/ebroecker/canmatrix/files/13042067/obd2-test.zip)

The expected behavior would be that canmatrix would correctly interpret the ID `2024` as a standard 11-bit ID, while the ID `2564485397` should be interpreted as a J1939 PGN based on the `vFrameFormat` attributes. This should in turn allow for decoding both via the same DBC.

ebroecker, 2023-10-19:
Thanks for this report, gonna have a look on it...

ebroecker, 2023-10-20:
Hi @MatinF ,

which version of canmatrix do you use?
I just tried to reproduce the issue, but was not successful...

```python
import canmatrix.formats

my_matrix = canmatrix.formats.loadp_flat(r"CSS-Electronics-OBD2-incl-extended-v2.0.dbc")

for num, frame in enumerate(my_matrix.frames):
    print(f"Frame {num}: {frame}")
    print(f" is j1939: {frame.is_j1939}")
    print(f" id:  {frame.arbitration_id}")
    print(f' Format: {frame.attributes["VFrameFormat"]}')
    if frame.is_j1939:
        print(f" pgn: {hex(frame.arbitration_id.pgn)}")
```

gives me this output:
```
Frame 0: OBD2
 is j1939: False
 id:  ArbitrationId(id=2024, extended=False)
 Format: StandardCAN
Frame 1: OBD2
 is j1939: True
 id:  ArbitrationId(id=417001749, extended=True)
 Format: J1939PG
 pgn: 0xda00
 ```

MatinF, 2023-10-22:
Hi Eduard,

This was tested via asammdf v7.3.16 GUI build on Windows 10 64 bit. I'm not entirely sure which canmatrix version is being used here. In this version, nothing is returned when I try to decode the test MF4 with the DBC. When I run a dev version with a console, that's where I see the previously reported error. 

@danielhrisca is it possible to try and update to the latest canmatrix in asammdf to see if this resolves this issue?

MatinF, 2023-10-24:
Thanks, I'll update on this in the separate ticket on the asammdf github https://github.com/danielhrisca/asammdf/issues/930

closed 2023-10-24 by MatinF