ebroecker/canmatrix #733: Error while adding xlsx support

NordishKiel, 2023-11-02:
Hi,
trying to add additional file support via "pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[xlsx]"
Python 3.12
git version 2.42.0.windows.2
pip 23.3.1 from C:\Python312\Lib\site-packages\pip (python 3.12)

Dump:
C:\Program Files\Git>pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[xlsx]
DEPRECATION: git+https://github.com/ebroecker/canmatrix#egg=canmatrix[xlsx] contains an egg fragment with a non-PEP 508 name pip 25.0 will enforce this behaviour change. A possible replacement is to use the req @ url syntax, and remove the egg fragment. Discussion can be found at https://github.com/pypa/pip/issues/11617
Collecting canmatrix[xlsx]
  Cloning https://github.com/ebroecker/canmatrix to c:\users\urlau\appdata\local\temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2
  Running command git clone --filter=blob:none --quiet https://github.com/ebroecker/canmatrix 'C:\Users\urlau\AppData\Local\Temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2'
  Resolved https://github.com/ebroecker/canmatrix to commit b4524e45157dc8374d9d99dab04a10f28201749f
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [31 lines of output]
      C:\Users\urlau\AppData\Local\Temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2\versioneer.py:418: SyntaxWarning: invalid escape sequence '\s'
        LONG_VERSION_PY['git'] = '''
      Traceback (most recent call last):
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\urlau\AppData\Local\Temp\pip-build-env-iddl7urg\overlay\Lib\site-packages\setuptools\build_meta.py", line 355, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\urlau\AppData\Local\Temp\pip-build-env-iddl7urg\overlay\Lib\site-packages\setuptools\build_meta.py", line 325, in _get_build_requires
          self.run_setup()
        File "C:\Users\urlau\AppData\Local\Temp\pip-build-env-iddl7urg\overlay\Lib\site-packages\setuptools\build_meta.py", line 507, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "C:\Users\urlau\AppData\Local\Temp\pip-build-env-iddl7urg\overlay\Lib\site-packages\setuptools\build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 83, in <module>
        File "C:\Users\urlau\AppData\Local\Temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2\versioneer.py", line 1477, in get_version
          return get_versions()["version"]
                 ^^^^^^^^^^^^^^
        File "C:\Users\urlau\AppData\Local\Temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2\versioneer.py", line 1409, in get_versions
          cfg = get_config_from_root(root)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\urlau\AppData\Local\Temp\pip-install-7ov8yrpt\canmatrix_9054a453bf0e48b9b4f3db4a4e218ff2\versioneer.py", line 339, in get_config_from_root
          parser = configparser.SafeConfigParser()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

ebroecker, 2024-07-18:
Hi @NordishKiel 

is this error still there?

NordishKiel, 2024-07-18:
Hi, i will check that soon.--Gesendet mit der WEB.DE Mail App

ebroecker, 2024-12-18:
nothing new here, please reopen if still an issue

closed 2024-12-18 by ebroecker