ebroecker/canmatrix #736: Arxml recognizes canfd as can

mominggame, 2023-11-22:
`(is_fd_elem is not None and is_fd_elem.text == 'TRUE'):`
I noticed that when using canmatrix.formats.loadp to parse an ARXML file, all instances of 'canfd' are being interpreted as 'can'. After examining the source code, I observed that the check is for the uppercase 'TRUE', while my ARXML file uses the lowercase 'true'.

ebroecker, 2023-11-22:
Hi @mominggame,

thanks for your hint.
I'll have a look on it and probably fix it.

ebroecker, 2023-11-22:
Hi @mominggame 

can you please check branch iss_736?

pip install git+https://github.com/ebroecker/canmatrix/tree/iss_736

mominggame, 2023-11-23:
Since version 1.0, I have been unable to parse large arxml files, and I'm not sure if it's due to the file size. Currently, the situation on my end is that some relatively small arxml files can be parsed, but large files get stuck in a loop, and the 'canmatrix.formats.loadp' function never stops. However, I did not encounter this issue in version 0.9.5. My workaround was to change '(is_fd_elem is not None and is_fd_elem.text == 'TRUE'):' to '(is_fd_elem is not None and is_fd_elem.text.upper() == 'TRUE'):' in version 0.9.5. My English is not good, and this passage has been translated using machine translation, so it may sound a bit awkward. I hope for your understanding.

mominggame, 2023-11-23:
Is the speed slowing down? I finally parsed the large file, but it took 30 minutes for the 80MB arxml!

ebroecker, 2023-11-23:
Hi @mominggame 

I'll investigate the speedissue - this should Not Happen ....

ebroecker, 2023-11-23:
Hi @mominggame,

I had a quick view over the changes.
Not so much changes in the arxml-code, but one change:
Gateway information is parsed now. 
So is your ARXML-File with gateway information? Just for excluding some possible errors....

closed 2024-05-23 by ebroecker