ebroecker/canmatrix #739: Convert arxml file to dbc file wrong

MaggieLxj, 2023-11-30:
when i convert the arxml file, I found signals in the file, but no dbc file output. Maybe it's because the size of the arxml file is 9M? Please help me solve this problem, thank you. 

└─[0] canconvert -vv 'H37_Project_SOMEIP_Service_Table_V7.0.1_TDA4_Acore-20231031023628.arxml' H37.dbc
INFO - convert - Importing H37_Project_SOMEIP_Service_Table_V7.0.1_TDA4_Acore-20231031023628.arxml ... 
DEBUG - arxml - Read arxml ...
DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 506 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - CONTAINER-I-PDU-HEADER-BYTE-ORDER not found (default big endian)
INFO - convert - done

INFO - convert - Exporting H37.dbc ... 
INFO - convert - done

Then nothing happened.

MaggieLxj, 2023-12-01:
Thank you for your answer.
Actually, I've used " --arxmlIgnoreClusterInfo" and "--arxmlEthernet" to generate dbc, and found some signals. But the signals I wanted did't in the dbc file, they were in the arxml file. 
So, should I convert all signals in the arxml file? How can I do?

MaggieLxj, 2023-12-01:
Thank you for your answer. I've already get the signals I want by adding "--arxmlEthernet" command line switch. 
The next question is how to get the dbc file in python script? Don't use the style of 'os.system(command line)', just use canmatrix API.

ebroecker, 2023-12-06:
hi @MaggieLxj 
canmatrix.formats.loadp() and friends should solve your issue

MaggieLxj, 2023-12-06:
Just pop-up window said that when I open the dbc file use canoe. No more information. Later I can take a screen shot and show you

---- Replied Message ----
| From | Eduard ***@***.***> |
| Date | 12/06/2023 20:22 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [ebroecker/canmatrix] Convert arxml file to dbc file wrong (Issue #739) |

Hi @MaggieLxj

what ist the "segmentation description" ?
Do you have some screenshot of the error?

ebroecker, 2024-06-19:
no new infos here...
please reopen if news available

closed 2024-06-19 by ebroecker