ebroecker/canmatrix #742: Python3.12 compatibility / drop 'future' dependency

umlaeute, 2023-12-12:
`canmatrix` depends on `future`, but it seems that `future` is not really compatible with Python3.12 (yet?).

One possible solution is to fix `future`, but
- this is more complicated than my average python foo
- `future` is a compat layer for Py2/Py3, but according to https://github.com/ebroecker/canmatrix/releases/tag/1.0, we've seen the "Last py2 release", so it's probably time to drop cruft.
- `canmatrix` uses a single function from `future`, which can trivially be implemented in ordinary Python
 

here's a patch that eliminates the use of `future`, but requires Python3:

```diff
--- python-canmatrix.orig/src/canmatrix/formats/sym.py
+++ python-canmatrix/src/canmatrix/formats/sym.py
@@ -100,11 +100,7 @@
 
 def create_signal(db, signal):  # type: (canmatrix.CanMatrix, canmatrix.Signal) -> str
     output = ""
-    if sys.version_info > (3, 0):
-        quote_name = not signal.name.isidentifier()
-    else:
-        from future.utils import isidentifier
-        quote_name = not isidentifier(signal.name)
+    quote_name = not signal.name.isidentifier()
     if quote_name:
         output += 'Var="%s" ' % signal.name
     else:
```

umlaeute, 2023-12-12:
a more complete solution would be to just drop all those occurences of `if sys.version_info <= (3,0)`

umlaeute, 2023-12-12:
also note that
> DeprecationWarning: The SafeConfigParser class has been renamed to ConfigParser in Python 3.2. This alias will be removed in Python 3.12. Use ConfigParser directly instead.

since Python3.12 has arrived, this needs to be handled somehow:

```diff
--- python-canmatrix.orig/versioneer.py
+++ python-canmatrix/versioneer.py
@@ -336,7 +336,7 @@
     # configparser.NoOptionError (if it lacks "VCS="). See the docstring at
     # the top of versioneer.py for instructions on writing your setup.cfg .
     setup_cfg = os.path.join(root, "setup.cfg")
-    parser = configparser.SafeConfigParser()
+    parser = configparser.ConfigParser()
     with open(setup_cfg, "r") as f:
         parser.readfp(f)
     VCS = parser.get("versioneer", "VCS")  # mandatory
```

this patch requires Python>=3.2 which is probably acceptable (or not)

ebroecker, 2023-12-13:
Hi @umlaeute ,

thanks for your help.

And yes, you are right.
I created 2 pull requests for your patches:
https://github.com/ebroecker/canmatrix/pull/744
https://github.com/ebroecker/canmatrix/pull/745

Thanks again

umlaeute, 2023-12-13:
sorry to have created additional work.

i could (and would) of course have done the PRs myself, but wanted to get your feedback first to see if i was on the right track (and it was easier for my actual workflow (while preparing the Debian packages) to just paste the diffs for now)

closed 2024-05-23 by ebroecker