ebroecker/canmatrix #750: Loading arxml with Ethernet data Performance is extremely bad on development branch(2023-12-15)

xRowe, 2023-12-15:
Hello

I tried to using this awesome lib to parse CAN/Ethernet pdu in realtime when receiving data.
Unlucky, my arxml is not able to parse succesfully on official v1.0, 
and I tried the latest code on development branch and the v0.95 one.

But v0.95 speed is much faster than the development branch.
 
below are the cProfile I run the script
`
import time
from canmatrix.formats.arxml import load
import cProfile

def load_arxml():
    arxml = "abc.arxml"
    start = time.perf_counter()
    dbc = load(arxml, decode_ethernet=True)
    end = time.perf_counter()
    print(f"Total: {end-start}s")

cProfile.run("load_arxml()")
`
[v095.log](https://github.com/ebroecker/canmatrix/files/13684794/v095.log)
[v1.log](https://github.com/ebroecker/canmatrix/files/13684795/v1.log)

ebroecker, 2023-12-15:
Hi @xRowe 

Thanks for the analysis.

I had already a look but did not yet find the patch witch made it worse. I'll keep on searching

xRowe, 2023-12-16:
Hello @ebroecker 

From my test, I found the below code takes much of time.
https://github.com/ebroecker/canmatrix/blob/d9902ace093fc1bb31af02599243847b529b0b44/src/canmatrix/formats/arxml.py#L1135C1-L1141C21

If I comment the code, the perfomance will be good.

Sorry, I am new on arxml, could not give some further investigate now

Thanks,
Rowe

ebroecker, 2023-12-18:
Hi @xRowe 
Thank you for this analysis. 
So I know where to look at...

ebroecker, 2023-12-20:
Hi  @xRowe,

could you try branch https://github.com/ebroecker/canmatrix/tree/issue_750

 pip install git+https://github.com/ebroecker/canmatrix/tree/issue_750
 
 Maybe this little fix speeds up the parsing again

ebroecker, 2023-12-23:
Hi @xRowe 

thank you for testing!
issue_750 should be created from development, but I just now did an update of the branch. It should now depend on current development branch.

But anyway, you tried it, and it seems to fix the time-issue (more or less).
I'll do some cleanup and then merge it into development.

commit 8fcc2829 (2023-12-30): ARXML parsing speed issue #750 (#751)

* may fix #750

* remove not longer needed list of signal_triggerings

* arxml: fix some type hints

---------
closed 2023-12-30 by ebroecker