ebroecker/canmatrix #753: use Canmatrix  create dbc

maxceng, 2023-12-28:
hello, i  set is_little_endian to false in python canmatrix.signal.
, but the byte order in the generated dbc is Motorola format, and the start bit is converted into intel format.how can I solve it,thanks

ebroecker, 2023-12-28:
Hi  @maxceng 

https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder:

little endian == Intel == MOST-SIGNIFICANT-BYTE-LAST
big endian == Motorola == MOST-SIGNIFICANT-BYTE-FIRST

is_litte_endian = False means "big endian" means Motorola 
which seems correct to me.

Could you provide an example?

maxceng, 2023-12-29:
I  read a signal from excel  such as  (startbit: 0  ,length  :4,byter order: motorola),I use  code canmatrix.Signal(        name=sig_name,        comments=sig_comment,        is_little_endian=false,)canmatrix.formats.dumpp({"": network}, "output.dbc", "dbc")write and export to dbc,but the start bit is changed to 4,but the byter order is still  motorola
-------- 原始信息 --------发件人： Eduard Bröcker ***@***.***> 日期: 2023/12/28  23:58  (GMT+08:00) 收件人： ebroecker/canmatrix ***@***.***> 抄送： YuukiZ ***@***.***>, Mention ***@***.***> 主题： Re: [ebroecker/canmatrix] use Canmatrix  create dbc (Issue #753) 
Hi  @maxceng
https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder:
little endian == Intel == MOST-SIGNIFICANT-BYTE-LAST
big endian == Motorola == MOST-SIGNIFICANT-BYTE-FIRST
is_litte_endian = False means "big endian" means Motorola
which seems correct to me.
Could you provide an example?

ebroecker, 2023-12-30:
Hi @maxceng 

I still cannot reproduce the issue.

Look at following code:

```python
import canmatrix

network = canmatrix.CanMatrix()

sig_name = "my_name"
sig_comment  = "my_comment"
sig_name2 = "my_name2"
signal = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=True)
signal2 = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=False)

network.add_signal(signal)
network.add_signal(signal2)

canmatrix.formats.dumpp({"": network}, "output.dbc", "dbc")
```
which produces:
```
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 3221225472 VECTOR__INDEPENDENT_SIG_MSG: 0 Vector__XXX
 SG_ my_name0 : 0|0@1- (1,0) [0|0] "" Vector__XXX
 SG_ my_name1 : 7|0@0- (1,0) [0|0] "" Vector__XXX
```

This seems ok for me: `@1` vs. `@0`.

Could you please provide more detail, what is going wrong at your side?
Thanks

maxceng, 2024-01-02:
hello ,for example:import canmatrixnetwork = canmatrix.CanMatrix()message = canmatrix.Frame(is_fd=True,name="msg_test",arbitration_id=0x1a1,cycle_time=10,size=64)frame = network.add_frame(message)signal = canmatrix.Signal(name="sig_name1",is_little_endian=False,start_bit=10,size=6)message.add_signal(signal)canmatrix.formats.dumpp({"":network}, "output.dbc", "dbc")print("done")output:VERSION "created by canmatrix"NS_ :BS_:BU_: BO_ 417 msg_test: 64 Vector__XXX SG_ sig_name1 : ***@***.*** (1,0) [-32|31] "" Vector__XXXBA_DEF_ BO_ "GenMsgCycleTime" INT 0 65535;BA_DEF_ BO_ "VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","StandardCAN_FD","ExtendedCAN_FD";BA_DEF_  "BusType" STRING;BA_ "BusType"   "CAN FD";BA_ "GenMsgCycleTime" BO_ 417 10;BA_ "VFrameFormat" BO_ 417 14;
-------- 原始信息 --------发件人： Eduard Bröcker ***@***.***> 日期: 2023/12/31  02:59  (GMT+08:00) 收件人： ebroecker/canmatrix ***@***.***> 抄送： YuukiZ ***@***.***>, Mention ***@***.***> 主题： Re: [ebroecker/canmatrix] use Canmatrix  create dbc (Issue #753) 
Hi @maxceng
I still cannot reproduce the issue.
Look at following code:
import canmatrix

network = canmatrix.CanMatrix()

sig_name = "my_name"
sig_comment  = "my_comment"
sig_name2 = "my_name2"
signal = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=True)
signal2 = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=False)

network.add_signal(signal)
network.add_signal(signal2)

canmatrix.formats.dumpp({"": network}, "output.dbc", "dbc")
which produces:
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 3221225472 VECTOR__INDEPENDENT_SIG_MSG: 0 Vector__XXX
 SG_ my_name0 : ***@***.*** (1,0) [0|0] "" Vector__XXX
 SG_ my_name1 : ***@***.*** (1,0) [0|0] "" Vector__XXX

This seems ok for me: @1 vs. @0.
Could you please provide more detail, what is going wrong at your side?
Thanks

maxceng, 2024-01-02:
hi,i have another question:can I input Chinese in comment?when I input "测试case",the comment only contains "case"
-------- 原始信息 --------发件人： Eduard Bröcker ***@***.***> 日期: 2023/12/31  02:59  (GMT+08:00) 收件人： ebroecker/canmatrix ***@***.***> 抄送： YuukiZ ***@***.***>, Mention ***@***.***> 主题： Re: [ebroecker/canmatrix] use Canmatrix  create dbc (Issue #753) 
Hi @maxceng
I still cannot reproduce the issue.
Look at following code:
import canmatrix

network = canmatrix.CanMatrix()

sig_name = "my_name"
sig_comment  = "my_comment"
sig_name2 = "my_name2"
signal = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=True)
signal2 = canmatrix.Signal(name=sig_name, comments=sig_comment, is_little_endian=False)

network.add_signal(signal)
network.add_signal(signal2)

canmatrix.formats.dumpp({"": network}, "output.dbc", "dbc")
which produces:
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 3221225472 VECTOR__INDEPENDENT_SIG_MSG: 0 Vector__XXX
 SG_ my_name0 : ***@***.*** (1,0) [0|0] "" Vector__XXX
 SG_ my_name1 : ***@***.*** (1,0) [0|0] "" Vector__XXX

This seems ok for me: @1 vs. @0.
Could you please provide more detail, what is going wrong at your side?
Thanks

ebroecker, 2024-01-02:
Hi @maxceng 

you need only to give the export-encoding:

`canmatrix.formats.dumpp({"":network}, "output.dbc", "dbc", dbcExportEncoding="utf8")`

maxceng, 2024-01-03:
thanks  ,this problem is solved
-------- 原始信息 --------发件人： Eduard Bröcker ***@***.***> 日期: 2024/1/3  05:16  (GMT+08:00) 收件人： ebroecker/canmatrix ***@***.***> 抄送： YuukiZ ***@***.***>, Mention ***@***.***> 主题： Re: [ebroecker/canmatrix] use Canmatrix  create dbc (Issue #753) 
Hi @maxceng
you need only to give the export-encoding:
canmatrix.formats.dumpp({"":network}, "output.dbc", "dbc", dbcExportEncoding="utf8")

closed 2024-06-19 by ebroecker