ebroecker/canmatrix #756: Convert DBC to arxml failed

hejiashou, 2024-01-04:
Traceback (most recent call last):
  File "/home/hjs/.local/bin/canconvert", line 8, in <module>
    sys.exit(cli_convert())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/cli/convert.py", line 158, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/convert.py", line 351, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/formats/__init__.py", line 119, in dumpp
    dump(can_cluster, file_object, export_type, **options)
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/formats/__init__.py", line 106, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/formats/arxml.py", line 800, in dump
    for value in sorted(signal.values, key=lambda x: int(x, 0)):
  File "/home/hjs/.local/lib/python3.10/site-packages/canmatrix/formats/arxml.py", line 800, in <lambda>
    for value in sorted(signal.values, key=lambda x: int(x, 0)):
TypeError: int() can't convert non-string with explicit base

pempem98, 2024-01-04:
@hejiashou  let try to use int(x) instead of int(x, 0)

pempem98, 2024-03-15:
> I am also experiencing this issue! When can we expect this to be resolved? With proper guidance I might also be able to fix this.
> 
> ```
> INFO - convert - done
> 
> INFO - convert - Exporting easy.arxml ...
> INFO - convert -
> INFO - convert - 2 Frames found
> Traceback (most recent call last):
>   File "/home/bijesh/Playground/python/can/venv/bin/canconvert", line 8, in <module>
>     sys.exit(cli_convert())
>              ^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
>     return self.main(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/click/core.py", line 1078, in main
>     rv = self.invoke(ctx)
>          ^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/click/core.py", line 783, in invoke
>     return __callback(*args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/cli/convert.py", line 158, in cli_convert
>     canmatrix.convert.convert(infile, outfile, **options)
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/convert.py", line 351, in convert
>     canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/formats/__init__.py", line 119, in dumpp
>     dump(can_cluster, file_object, export_type, **options)
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/formats/__init__.py", line 106, in dump
>     module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
>     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/formats/arxml.py", line 800, in dump
>     for value in sorted(signal.values, key=lambda x: int(x, 0)):
>                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/bijesh/Playground/python/can/venv/lib/python3.11/site-packages/canmatrix/formats/arxml.py", line 800, in <lambda>
>     for value in sorted(signal.values, key=lambda x: int(x, 0)):
>                                                      ^^^^^^^^^
> TypeError: int() can't convert non-string with explicit base```
> ```

Can I have the input DBC of this case?

ebroecker, 2024-07-18:
should be merged

closed 2024-07-18 by ebroecker