ebroecker/canmatrix #760: Renaming Node / ECU via rename_ecu should clear out any existing transmitter and receiver

ChristianSteffens, 2024-01-10:
Hi,

I noticed a problem (bug?) in the `db.rename_ecu` [method](https://github.com/ebroecker/canmatrix/blob/ef38e9403598c6a667cbb39f980b92e6258eac6d/src/canmatrix/canmatrix.py#L2079) and specifically the usage of the `frame.update_receiver()` [method](https://github.com/ebroecker/canmatrix/blob/ef38e9403598c6a667cbb39f980b92e6258eac6d/src/canmatrix/canmatrix.py#L1286).

The method should cover a rename attempt of an existing ECU throughout the complete database: 
- Any existing usage of an ECU with the old name is removed from frames transmitter and signals receiver list - works as expected.
- Then it calls `frame.update_receiver()` to make sure the frame's signal receiver list is propagated to the frame itself.

But here a catch as the frame's receiver list is not cleared / emptied before (!) updating the list, so any previous occurance of any ECU (e.g. the ECU with the old name) is still carried over.

So the correct method should empty the receiver array of the frame before updating it via the signals receiver list.

ChristianSteffens, 2024-05-21:
Any interest in fixing this - if not should I raise a PR?

ebroecker, 2024-06-19:
Hi @ChristianSteffens ,

for sure, yes, feel free to raise an PR

ebroecker, 2024-07-18:
should be merged

closed 2024-07-18 by ebroecker