ebroecker/canmatrix #762: when xsl2dbc,singal's offset 、mini  is uncorrect

wangzheng0902, 2024-01-15:
version：1.0
when change a xls file to dbc file.All singals' offset and mini is 0,not value in xls.

ebroecker, 2024-07-18:
please try `--additionalSignalAttributes min,max` during excel-export

notmmao, 2024-12-02:
my test script:
```sh
FN=test_db

canconvert ${FN}.dbc ${FN}.xlsx --dbcImportEncoding GBK --dbcImportCommentEncoding GBK --additionalSignalAttributes min,max,offset,factor
canconvert ${FN}.xlsx ${FN}.xlsx.dbc --dbcExportEncoding GBK --dbcExportCommentEncoding GBK
cancompare ${FN}.dbc ${FN}.xlsx.dbc > ${FN}.xlsx.log
```

min,max,offset work as expect, but **factor** not.
In the ${FN}.xlsx.log file, appears
```
  FRAME changed  XXX_2
    SIGNAL changed  XXX_2_YYY
      factor changed  XXX_2_YYY
      <class 'decimal.Decimal'>
old: 0.001953125 new: 0.00195312
```
In the ${FN}.xlsx
```
# pseudocode..
- col[Function / Increment Unit] == 0.00195312  V
- col[signal.factor] == 0.001953125
```
It seems that `signal.factor` has **lower** priority than `Function / Increment Unit` in the load function.
I made some modifications to the code to meet my expectations.
```python
# xls.load function, about line:552
factor = sh.cell(row_num, index['function']).value
if isinstance(factor, past.builtins.basestring):
	factor = factor.strip()
	if " " in factor and factor[0].isdigit():
		(factor, unit) = factor.strip().split(" ", 1)
		factor = factor.strip()
		unit = unit.strip()
		new_signal.unit = unit
		try:
		    # only update when factor is default
			if new_signal.factor in (1, 1.0):		  # <== here
				new_signal.factor = float_factory(factor)
		except:
			logger.warning(
				"Some error occurred while decoding scale of Signal %s: '%s'",
				signal_name,
				sh.cell(row_num, index['function']).value)
	else:
		unit = factor.strip()
		new_signal.unit = unit
		new_signal.factor = 1
```

ebroecker, 2024-12-18:
Hi  @notmmao 

thanks,
I merged it in [pr](https://github.com/ebroecker/canmatrix/pull/824)

closed 2025-01-15 by ebroecker