ebroecker/canmatrix #766: convert ARXML to DBC have error:<class 'decimal.DivisionUndefined'>

mrmuyi, 2024-02-15:
I want to convert arxml to dbc, the cmd as "canconvert -vv ess3337.arxml ess377.dbc", log as below, could you kindly help?
python version 3.8.5
canmatrix version 1.0

DEBUG - arxml - PDU: Tester_Sporadic_AtifRequestRFSDA
DEBUG - arxml - processing Frame-Trigger: Tester_BKPCANFD_AtifRequestRHRDA
DEBUG - arxml - Frame: Tester_BKPCANFD_AtifRequestRHRDA
DEBUG - arxml - PDU: Tester_Sporadic_AtifRequestRHRDA
Traceback (most recent call last):
  File "C:\Program Files\Python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python38\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python38\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\Python38\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\Python38\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\Python38\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\cli\convert.py", line 155, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 2005, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1923, in decode_can_helper
    sig_value_hash[sig.name] = sig.phys2raw()
  File "C:\Program Files\Python38\lib\site-packages\canmatrix\canmatrix.py", line 450, in phys2raw
    raw_value = (self.float_factory(value) - self.float_factory(self.offset)) / self.float_factory(self.factor)
decimal.InvalidOperation: [<class 'decimal.DivisionUndefined'>]

ebroecker, 2024-02-19:
Hi @mrmuyi 

maybe your Frame "Tester_Sporadic_AtifRequestRHRDA" does not have a compu-method or the compu-method is incomplete.
Canmatrix fails because it does not find "factor" in the scaling of "Tester_Sporadic_AtifRequestRHRDA"

mrmuyi, 2024-02-22:
Hi, thanks for your quick reply!
in order to find where is the 0 factor, i print the message by "print(self.name)..."
question 1: I got the result as follow which is not match in VECTOR tool or notebook

then i try to filter out the error by "if self.factor==0  return 0"
and got another error: 'Nonetype' object has no attribute "soureceline"

[image]
[image]
"
[image]
[image]
[image]

mrmuyi, 2024-02-22:
Hi, thanks for your quick reply!
in order to find where is the 0 factor, i print the message by "print(self.name)..."
question 1: I got the result as follow which is not match in VECTOR tool or notebook

then i try to filter out the error by "if self.factor==0 return 0"
and got another error: 'Nonetype' object has no attribute "soureceline"
[image]
[image]
[image]
[image]
[image]

ebroecker, 2024-06-19:
Hi @mrmuyi 

is this bug still there or maybe it is fixed with https://github.com/ebroecker/canmatrix/pull/776 already?

ebroecker, 2024-12-18:
please reopen, if still occures

closed 2024-12-18 by ebroecker