ebroecker/canmatrix #767: Convert dbc <=> xlsx - units are deleted

nexulm, 2024-02-16:
During the conversion of dbc to xls and vice versa all units of signals are deleted.

**Input;** 
Standard-CAN dbc

**Command used with Ubuntu Linux 22.04:**
python3 -m canmatrix.cli.convert "$path/$param_file_wo_extension.dbc" "$path/$param_file_wo_extension.xlsx"

Is there any option I've missed to include the units as well?
Since I'm using Canmatrix to convert the DBC, edit the entries in Excel and create an updated DBC from it, the observed behavior is not helpful.

ebroecker, 2024-02-19:
Hi @nexulm,

xls(x) is a lossy format, it will never keep all information from dbc.

But indeed, you can add additional information with:
`–additionalSignalAttributes//–additionalFrameAttributes`

ebroecker, 2024-06-20:
Hi @nexulm 

forget the upper comment, it was nonsense

This should be fixed now also, was another bug...

nexulm, 2024-06-21:
Fixed, not reproducable anymore since commit 8b751faab12d590de316cc5f7b43b2f43680c506