ebroecker/canmatrix #768: Convert dbc <=> xlsx - DLC is always reset to it's maximum of 8 bytes

nexulm, 2024-02-16:
During the conversion of dbc to xls and vice versa all CAN messages are resetted to max DLC of 8 bytes.

Input;
Standard-CAN dbc

Command used with Ubuntu Linux 22.04:
python3 -m canmatrix.cli.convert "$path/$param_file_wo_extension.dbc" "$path/$param_file_wo_extension.xlsx"

Is there any option I've missed to avoid resetting the current DLC length (i.e. DLC =3) to the max possible DLC bytes of 8?
Since I'm using Canmatrix to convert the DBC, edit the entries in Excel and create an updated DBC from it, the observed behavior is not helpful.

ebroecker, 2024-02-19:
same here:
xls(x) is lossy, but you can add additional Attributes:
 `–additionalSignalAttributes//–additionalFrameAttributes`

closed 2024-06-19 by ebroecker

nexulm, 2024-06-19:
Is ist really close if the code ha a fixed value of dlc = 8 at line 409 of xlsx.py ???
[image]
---
I've fixed this specific issue in my fork by use frame.size as input.

ebroecker, 2024-06-20:
Hi @nexulm,

you are right, the root-cause is different - all additional attributes are not read anymore since I switched to openpyxl.
Will be fixed in #541 (which I just reopened)