ebroecker/canmatrix #772: Exporting ARXML to XLSX (CanCluster vs. J1939Cluster)

ganralf75, 2024-02-23:
Hello,

thanks a lot for providing such a useful tool :-)

I have recognized a problem w/ a arxml file where I wanted to convert it to a more readable Excel document and I wanted to ask if that is in principal not possible or if it is a known/new issue.

I have converted a arxml that is defined as "CanCluster" and conversion looks pretty fine (xlsx document generated).
[image]

And then I converted a arxml file that is defined as "J1939Cluster" and conversion succeeded but there was no xlsx document generated.
[image]

Do you have an idea why the J1939 arxml does not generate any xlsx file?

thanks a lot in advance!

ebroecker, 2024-02-25:
Hi @ganralf75 ,

I never had a J1939-ARXML - so quite possible that nothing works.

Maybe you have some example you could provide?

ganralf75, 2024-03-25:
Hi @ebroecker,

wanted to ask if you already had the chance to analyze the problem? :-)

thanks!
ganralf75

ebroecker, 2024-03-26:
Hi @ganralf75 

sorry - not yet - but not forgotten !

ebroecker, 2024-03-26:
Hi @ganralf75,

I did a simple `python3 -m canmatrix.cli.convert --arxmlIgnoreClusterInfo J1939ClusterExample.arxml aa.xlsx`

which gave me at least this xlsx:
[aa.xlsx](https://github.com/ebroecker/canmatrix/files/14761137/aa.xlsx)

Maybe this already helps you?

I'll have to dive deeper into it, to create some dedicated support for J1939,...

ganralf75, 2024-04-09:
Hi @ebroecker,

thanks a lot for your feedback. Solution seems to work and xlsx is now created!

closed 2024-05-23 by ebroecker