ebroecker/canmatrix #773: canmatrix.py

daestx, 2024-02-23:
I have an issue with doing a diff on two DBC files. I get following debug info:

 File "canmatrix\formats\__init__.py", line 79, in loadp_flat
  File "canmatrix\formats\__init__.py", line 71, in loadp
  File "canmatrix\formats\__init__.py", line 90, in load
  File "canmatrix\formats\dbc.py", line 960, in load
  File "canmatrix\canmatrix.py", line 450, in phys2raw
decimal.InvalidOperation: [<class 'decimal.DivisionUndefined'>]

coming from this line
raw_value = (self.float_factory(value) - self.float_factory(self.offset)) / self.float_factory(self.factor)

obviously, division by 0 is not handled correctly.

I changed the code as workaorund

    if self.factor == 0:
        self.factor = 1
    
    raw_value = (self.float_factory(value) - self.float_factory(self.offset)) / self.float_factory(self.factor)

please check if there is a better solution

daestx, 2024-02-23:
issue  #766 looks similar

ebroecker, 2024-02-25:
Some Idea:

canmatrix for now only uses `COMPU-INTERNAL-TO-PHYS`  compu-methods, 

Maybe your ARXML uses `COMPU-PHYS-TO-INTERNAL`.

daestx, 2024-02-28:
I used dbc file as input not an ARXML

ebroecker, 2024-02-29:
Oh sorry you are right...

Do you have a chance to check the dbc itself? 
A factor of 0 does not make sense at all. I just want to know if it's a problem with canmatrix or with the dbc file...

daestx, 2024-03-04:
So, I checked the DBC file now and identified, that the DBC is buggy. According the documentation in https://www.influxtechnology.com/post/understanding-can-dbc  

[image]

So, I presume this definition of a DBC file I received is wrong, right?

[image]

So the question to me is if such an issue shall be handled in the library

ebroecker, 2024-03-04:
Hi @daestx 

thanks for your analysis.
Yes the dbc defines the scale (=factor) to zero which is nonsense.
Seems so, that candb++ has some plausibility-check here to get around with a scale of 0.
So maybe canmatrix should do the same...

ebroecker, 2024-05-23:
should be merged