ebroecker/canmatrix #778: Casting the frame id from scientific number format fails

djdesant, 2024-03-21:
db.enum_attribs_to_values()
  File "C:\Prj\MBD\Projects\BX\Hubwerk\GroundSensingRCP\ControlDesk\EmrGndSensing\Exp1\Measurement Data\MdfMetaData\.venv\Lib\site-packages\canmatrix\canmatrix.py", line 2317, in e
num_attribs_to_values
    frame.attributes[define] = self.frame_defines[define].values[int(frame.attributes[define])]
                                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '8.52715012702708E-317'

Please consider the casting to float before int. Example:
frame.attributes[define] = self.frame_defines[define].values[int(float(frame.attributes[define]))]

pempem98, 2024-03-23:
I guest this is the precision issue of the library **decimal**.
I used to get this issue before when I tried to read string '1' and convert it into decimal.Decimal type.
The result comes as decimal.Decimal('1.00000000000000000000003216...')

djdesant, 2024-04-10:
> I guest this is the precision issue of the library **[decimal](https://docs.python.org/3/library/decimal.html)** I used to get this issue before when I tried to read string '1' and convert it into decimal.Decimal type. The result comes as decimal.Decimal('1.00000000000000000000003216...')

Are there more possible workarounds ?
Could you please describe your alternative proposal for solving this problem ?

ebroecker, 2024-04-11:
I don't know a better workaround.
Also - I was not able to reproduce the error.

`pip install git+https://github.com/ebroecker/canmatrix.git@issue_778`

djdesant, 2024-04-21:
@ebroecker, when are planning to merge the current branch into master ?

ebroecker, 2024-05-23:
should be merged, if not please blame me

closed 2024-05-23 by ebroecker