ebroecker/canmatrix #781: DBC <=> XLSX initial_value

nexulm, 2024-04-11:
InitValue is transmitted to .xslx by converting .dbc to .xlsx with parameter:
--additionalSignalAttributes initial_value

How can I get the InitialValue back from the generated .xlsx row "signal.initial_value" back to the dbc?

So far, it seems for me that this functionality isn't integrated in the code?

ebroecker, 2024-04-11:
Hi @nexulm 

it is implemented, but maybe it does not work.
Have to have a look on it

nexulm, 2024-06-21:
It seems that the existing additional column "signal.initial_value" of the xlsx source file isn't used. Initial_value is always "0", only a few inital_value which are depending on the offset are showing another value than "0". But in this cases the offset calculation is the seems to be the reason to have the same value (initial_value = offset). For all signals the converted dbc shows initial_value = offset although the xlsx source has different values in the corresponding columns (signal.iniital_value and signal.offset)
The script is called with these parameters:

`python3 -m canmatrix.cli.convert --dbcImportEncoding=utf-8 \`
`--dbcImportCommentEncoding=utf-8 \`
`--dbcExportEncoding=utf-8 \`
`--dbcExportCommentEncoding=utf-8 \`
`--additionalSignalAttributes is_signed,factor,offset,min,max,initial_value,GenSigStartValue \`
`--additionalFrameAttributes size,DiagRequest,DiagResponse,ApplMessage,NmAsrMessage,GenMsgSendType \`
`"$path/$param_file_wo_extension.xlsx" "$path/$param_file_wo_extension.dbc"`

ebroecker, 2024-06-27:
Hi @nexulm 

could be fixed in branch: https://github.com/ebroecker/canmatrix/tree/issue_781

Additional: You don't need '--additionalSignalAttributes ... --additionalFrameAttributes ...' while **importing** xlsx files, they are only needed for export.

ebroecker, 2024-07-18:
should be merged

closed 2024-07-18 by ebroecker