ebroecker/canmatrix #786: Unnecessary Integer cast causes error during ARXML export

ChristianSteffens, 2024-05-06:
I might be mistaken, but isn't this integer conversion unnecessary? It causes an error during ARXML export, considering that the signal values are already stored as integers.

https://github.com/ebroecker/canmatrix/blob/8ca6d787e5c3faceac273fb00f35204fd6ad300f/src/canmatrix/formats/arxml.py#L802

```
canmatrix\formats\arxml.py", line 800, in <lambda>
    for value in sorted(signal.values, key=lambda x: int(x, 0)):
                                                     ^^^^^^^^^
TypeError: int() can't convert non-string with explicit base
```

ebroecker, 2024-06-27:
Hi @ChristianSteffens 

after a quick view to the code it seems you are right.

ebroecker, 2024-07-18:
should be merged

closed 2024-07-18 by ebroecker