ebroecker/canmatrix #797: Physical Conversion in XLSX Document

ganralf75, 2024-06-08:
Hello,

I have a stupid question regarding generation of XLSX document out of a arxml / dbc.

I have considered that when a CAN signal contains a value table (e.g. w/ 2 values ERR & SNA) the physical range (e.g. 0...1000km) of the signal is not visible in XLSX document. Instead only the values of the value table will be documented.

Is there any parameter that I can use so that the physical signal range will also be documented?

thanks in advance!
ganralf75

ganralf75, 2024-06-21:
thanks a lot @ebroecker! It works :-)

ganralf75, 2024-06-22:
Hi @ebroecker! I tried --additionalSignalAttributes min,max w/ arxml file and it seems that the min and max values are not plausible. Is it a bug or did I someting wrong?

In this example signal.max should be 25.5barg
[image]

kind regards,
ganralf75

ebroecker, 2024-06-23:
Hi @ganralf75 
not easy to say without the sources...

You could additionally add `signal.factor` and `signal.offset`  - so we can see what's calculated

ganralf75, 2024-06-26:
Hi @ebroecker, I added factor and offset. Both seem to be okay. 

[image]

I looks like the max is calculated from physical max (which is 25barg) and not from raw max (which is 250).

Signal configuration from Vector AUTOSAR Explorer
[image]

Signal Debugging Information
[image]

ebroecker, 2024-06-27:
Hi @ganralf75 
maybe the arxml is parsed the wrong way.
We had some issue using 'random' physical or internal contraints from 'ARXML'.

Maybe this bug is fixed with https://github.com/ebroecker/canmatrix/pull/779/files - but is not fixed right?

Maybe you could extract the contraints for this signal out of the ARXML, so we can isolate the problem ?

pempem98, 2024-09-07:
Look like the physical min/max was calculated twice, it would be better if we have a sample input to debug

ebroecker, 2024-12-18:
Hi @ganralf75 

do you have some example-input for us? Otherwise we are not able to reproduce.

ebroecker, 2026-11-16:
should be fixed, please reopen if still an issue

closed 2026-11-16 by ebroecker