ebroecker/canmatrix #806: Arxml convert to DBC

spark-0529, 2024-06-27:
hello here I want to vonvert the Arxml to DBC file ,but I find this issue ,please have a check ,thank you 
DEBUG - arxml - processing Frame-Trigger: FrTrEpbChassisCANFD2Fr01
DEBUG - arxml - Frame: EpbChassisCANFD2Fr01
DEBUG - arxml - PDU: EpbChassisCANFD2SignalIpdu01
DEBUG - arxml - get_signals: found I-SIGNAL-GROUP
DEBUG - arxml - get_signals: found I-SIGNAL-GROUP
Traceback (most recent call last):
  File "C:\python\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\python\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\python\lib\site-packages\canmatrix\cli\convert.py", line 160, in <module>
    sys.exit(cli_convert())
  File "C:\python\lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "C:\python\lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\python\lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\python\lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\python\lib\site-packages\canmatrix\cli\convert.py", line 155, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\python\lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\python\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\python\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\python\lib\site-packages\canmatrix\formats\arxml.py", line 2070, in load
    source_signal = ea.selector(source_triggering[0], "/I-SIGNAL-REF")
  File "C:\python\lib\site-packages\canmatrix\formats\arxml.py", line 306, in selector
    return sorted(result_list, key=lambda element: element.sourceline)
  File "C:\python\lib\site-packages\canmatrix\formats\arxml.py", line 306, in <lambda>
    return sorted(result_list, key=lambda element: element.sourceline)
AttributeError: 'NoneType' object has no attribute 'sourceline'

spark-0529, 2024-06-27:
[image]

spark-0529, 2024-06-27:
[image]

spark-0529, 2024-06-27:
[image]

spark-0529, 2024-07-09:
now it worked and successful convert ，thank you

spark-0529, 2024-07-09:
convert successful

closed 2024-07-09 by spark-0529

xuechengz0801, 2024-07-15:
> ![image](https://private-user-images.githubusercontent.com/80455550/343621452-c8d090cf-7a7a-4fc5-98fd-7ab7db477f54.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEwMTQwNDIsIm5iZiI6MTcyMTAxMzc0MiwicGF0aCI6Ii84MDQ1NTU1MC8zNDM2MjE0NTItYzhkMDkwY2YtN2E3YS00ZmM1LTk4ZmQtN2FiN2RiNDc3ZjU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE1VDAzMjIyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIxZDlmZThlNzlhZmI4MzBlMTNkNTljYTBlYWM0MGY5NjFmODc0OGVmNDhhYWZlYTU1ODZmZTMyMjI4MDdiYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tQ2IUyTT8cguFUPNcEm5qkrwGltzfRh_oVTgA3MC07k)

hi,spark-0529
  I have a same issue as yours,how to convert the .arxml even though I used 'python -m canmatrix.cli.convert...'?