ebroecker/canmatrix #821: start bit not correct when byte order is motorola

mzotw, 2024-11-19:
[image]

As pic above, 3 signals are defined with motorola format.

**checksum** starts from **bit 8**, length 16, 

**crashTyp** starts from **bit 123**, length 5,

but canmatrix parses as below
[image]
[image]

both start bits are incorrect

mzotw, 2024-11-19:
This happens only if byte order is motorola. With Intel format everything works ok.

ebroecker, 2024-11-26:
Hi @mzotw ,
This is the display-format of candb++ which is ab bit wired...

some test-dbc: 
[image]

if you just switch the display-format of candb++:

[image]

you can canmatrix make give you the same bit-numeration like in (default)candb++ with the signal-method `get_startbit(True,True)`:

```python
import canmatrix.formats

cm = canmatrix.formats.loadp_flat("your.dbc")
for index in range(3):
    print(cm.frames[0].signals[index].name + " " + str(cm.frames[0].signals[index].get_startbit(True, True)))
...
FlashTyp 123
_RolingCounter 16
_Checksum 8
```

mzotw, 2024-11-27:
@ebroecker  Thanks for your reply. That works for me.

closed 2024-11-27 by mzotw