ebroecker/canmatrix #829: For Motorola format in DBC, after use ccanmatrix.formats.loadp function to load, the start_bit are not right

CXZ1988, 2025-01-02:
Based on the Vector_DBC_File_Format_Documentation PDF, there is one point:
start_bit = unsigned_integer ;
The start_bit value specifies the position of the signal within the data field of the frame. 
For signals with byte order Intel (little endian) the position of the least-significant bit is given. 
**For signals with byte order Motorola (big endian) the position of the most significant bit is given.**

For Motorola format this is important to export it into excel.
See below example

CXZ1988, 2025-01-02:
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: Bot VCU EPS VDC VIU

BO_ 134 Bot_ExtBrkReq: 16 Bot
 SG_ BotBotBrkPedlPrsd : 5|1@0+ (1,0) [0|1] ""  VCU,VDC
 SG_ BotBotBrkPedlPrsdVldy : 7|2@0+ (1,0) [0|3] ""  VCU,VDC
 SG_ BotBotPedlTrvlOutpDrvr : 15|12@0-  (0.015625,-5) [-5|47] "mm"  VCU,VDC
 SG_ BotBotBrkPedlTrvlOutp : 19|12@0+ (0.015625,-5) [-5|47] "mm"  VCU,VDC
 SG_ BotBotBrkPedlTrvlOutpVldy : 33|2@0+ (1,0) [0|3] ""  VCU,VDC
 SG_ BotBotPedlTrvlOutpDrvrVldy : 35|2@0+ (1,0) [0|3] ""  VCU,VDC
 SG_ BotBotExtReqSts : 41|2@0+ (1,0) [0|3] ""  VCU,VDC
 SG_ BotBotExtReqPrioActv : 42|1@0+ (1,0) [0|1] ""  VCU,VDC
 SG_ Bot_ExtBrkReq_Cntr : 115|4@0+ (1,0) [0|15] ""  VCU,VDC
 SG_ Bot_ExtBrkReq_CRC : 127|8@0+ (1,0) [0|255] ""  VCU,VDC

BO_ 615 BotBotSts: 8 Bot
 SG_ BotBotPMaxSuprt : 7|8@0+ (1,0) [0|250] "bar"  VCU,VIU,VDC
 SG_ BotBotPFBlendingPotl : 15|8@0+ (0.5,0) [0|125] "bar"  VCU,VIU,VDC
 SG_ BotBotVCUCmpbltyIdx : 19|4@0+ (1,0) [0|15] ""  VCU,VIU,VDC
 SG_ BotBotTarStgyMod : 22|3@0+ (1,0) [0|7] ""  VCU,VIU,VDC
 SG_ BotBotDiagActv : 23|1@0+ (1,0) [0|1] ""  VCU,VIU,VDC
 SG_ BotBotPFBlendingPotlVldy : 29|2@0+ (1,0) [0|3] ""  VCU,VIU,VDC
 SG_ BotBotPMaxSuprtVldy : 31|2@0+ (1,0) [0|3] ""  VCU,VIU,VDC
 SG_ BotBotHbcReq : 36|1@0+ (1,0) [0|1] ""  VCU,VIU,VDC
 SG_ BotBotSts_Cntr : 51|4@0+ (1,0) [0|15] ""  VCU,VIU,VDC
 SG_ BotBotSts_CRC : 63|8@0+ (1,0) [0|255] ""  VCU,VIU,VDC

BO_ 136 VCU_TqRequest: 16 VCU
 SG_ VCU_PrpnTqReqDas : 7|16@0+ (1,-32768) [-32768|32766] "Nm" Vector__XXX
 SG_ VCU_BrkTqRgnReqQlfr : 17|2@0+ (1,0) [0|3] "" Vector__XXX
 SG_ VCU_TarGearReqVCU : 21|4@0+ (1,0) [0|15] "" Vector__XXX
 SG_ VCU_PrpnTqReqDasQlfr : 26|3@0+ (1,0) [0|7] "" Vector__XXX
 SG_ VCU_BrkTqRgnReq : 55|16@0+ (1,-32766) [-32766|0] "Nm" Vector__XXX
 SG_ VCU_TqReqStabyFA : 87|15@0+ (1,-16384) [-16384|16382] "Nm" Vector__XXX
 SG_ VCU_TqReqStabyRA : 103|15@0+ (1,-16384) [-16384|16382] "Nm" Vector__XXX
 SG_ VCU_TqReqStabyQlfrRA : 117|2@0+ (1,0) [0|3] "" Vector__XXX
 SG_ VCU_TqReqStabyQlfrFA : 119|2@0+ (1,0) [0|3] "" Vector__XXX
[... 540 lines cut ...]
VAL_ 695 VCU_BrkgCtrlTypEmgyAvl 0 "NotAvailable" 1 "FailureDetectedReducedAvl" 2 "FailureDetectedFullyAvl" 3 "Available" ;
VAL_ 695 VCU_BrkgCtrlTypCmftAvl 0 "NotAvailable" 1 "FailureDetectedReducedAvl" 2 "FailureDetectedFullyAvl" 3 "Available" ;
VAL_ 695 VCU_BrkgActvVehHldFlg 0 "False" 1 "True" ;
VAL_ 342 EPS_SteerWhlAgFailrStsEPS 0 "workingnormally" 1 "failure" ;
VAL_ 342 EPS_SteerWhlAgCalStsEPS 0 "Sensornotcalibrated" 1 "Sensoriscalibrated" ;
VAL_ 342 EPS_SteerWhlAgEps 16383 "Init" ;
VAL_ 145 VDC_CarWshngModActvnSt 0 "inactive" 1 "active" ;
VAL_ 145 VDC_BrkPedlFeelModReqd 0 "none" 1 "Normal" 2 "ECO" 3 "SPORT" ;
VAL_ 145 VDC_AccrPosnAct 254 "Abnormal" 255 "invalid" ;
VAL_ 145 VDC_LoglGearStsActQlfr 0 "Normal" 1 "Restricted" 2 "Invalid" 3 "Reserved" ;
VAL_ 145 VDC_AccrPosnActVldy 0 "invalid" 1 "valid" ;
VAL_ 145 VDC_PtSt 0 "Init/Afterrun" 1 "Standby" 2 "HV Ready" 3 "Drive/Remote Drive" 4 "Charging" 5 "Diagnostic" 6 "Reserved" 7 "Reserved" ;
VAL_ 145 VDC_LoglGearStsAct 0 "Init" 1 "Park" 2 "Reverse" 3 "Neutral" 4 "Drive" 5 "Reserved" 6 "Reserved" 7 "Error" 8 "Reserved" 9 "Reserved" 10 "Reserved" 11 "Reserved" 12 "Reserved" 13 "Reserved" 14 "Reserved" 15 "Reserved" ;
VAL_ 145 VDC_ExtEpbReq 0 "no request" 1 "close request" ;
VAL_ 145 VDC_PrpnTqReqDasAvl 0 "Not Available" 1 "Available" ;
VAL_ 145 VDC_PrpnTqDasOvrd 0 "Override not active" 1 "Override active" ;
VAL_ 145 VDC_ExactLoglGearStsAct 0 "Init" 1 "Park" 2 "Reverse" 3 "Neutral" 4 "Drive" 5 "Request_Park" 6 "Release_Park" 7 "Park_Epb_Engaged" 8 "Error" 9 "Epb_Engaged" 10 "ReqPark_VehMovg" 11 "Reserved" 12 "Reserved" 13 "Reserved" 14 "Reserved" 15 "Reserved" ;
VAL_ 145 VDC_VirtAccrPedlPosnVldy 0 "invalid" 1 "valid" ;
VAL_ 145 VDC_VirtAccrPedlPosn 254 "Abnormal" 255 "invalid" ;
VAL_ 605 IVI_AVN_AvhDrvrReq 0 "Norequest" 1 "Deactivate" 2 "Activate" 3 "Reserved" ;
VAL_ 605 IVI_AVN_HdcDrvrReq 0 "Norequest" 1 "Deactivate" 2 "Activate" 3 "Reserved" ;
VAL_ 605 IVI_AVN_VCUModReq 0 "VCUOn" 1 "VCUOff" ;
VAL_ 353 VDC_DrvrLimdByAdasLimr 0 "false" 1 "true" ;
VAL_ 353 VDC_AccrTqDmdAct 32767 "Init/Error" ;
VAL_ 353 VDC_CoastRgnTq 254 "Init" 255 "Error" ;

CXZ1988, 2025-01-02:
{'': CanMatrix(type=<matrix_class.CAN: 1>, attributes={'DBName': 'VCU_Matrix'}, ecus=[Ecu(name='Bot', comment=None), Ecu(name='VCU', comment=None), Ecu(name='EPS', comment=None), Ecu(name='VDC', comment=None), Ecu(name='VIU', comment=None)], frames=[Frame(name='Bot_ExtBrkReq', arbitration_id=ArbitrationId(id=134, extended=False), size=16, transmitters=['Bot'], is_complex_multiplexed=False, is_fd=False, comment='å\x88¶å\x8a¨ç\x94¨ç\x94µå\xad\x90å\x8a©å\x8a\x9bæ³µå¤\x96é\x83¨å\x88¶å\x8a¨è¯·æ±\x82\nBot external brake request', signals=[Signal(name='BotBotBrkPedlPrsd', start_bit=2, size=1, is_little_endian=False, is_signed=False, offset=Decimal('0'), factor=Decimal('1'), unit='', receivers=['VCU', 'VDC'], comment='è¡¨å¾\x81é©¾é©¶å\x91\x98è¸©å\x88¶å\x8a¨è¸\x8fæ\x9d¿ç\x9a\x84ä¿¡å\x8f·\nSignal that corrVCUonds to the information if the driver is applying the brake pedal', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={0: 'False', 1: 'True'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('0'), min=Decimal('0'), max=Decimal('1')), Signal(name='BotBotBrkPedlPrsdVldy', start_bit=0, size=2, is_little_endian=False, is_signed=False, offset=Decimal('0'), factor=Decimal('1'), unit='', receivers=['VCU', 'VDC'], comment='é©¾é©¶å\x91\x98å\x88¶å\x8a¨è¸\x8fæ\x9d¿ä¿¡å\x8f·æ\x9c\x89æ\x95\x88æ\x80§\nQualifier for sigEbbBrkPedlPrsd', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={0: 'Not initialized', 1: 'Normal', 2: 'Faulty', 3: 'Reserved'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('0'), min=Decimal('0'), max=Decimal('3')), Signal(name='BotBotPedlTrvlOutpDrvr', start_bit=8, size=12, is_little_endian=False, is_signed=True, offset=Decimal('-5'), factor=Decimal('0.015625'), unit='mm', receivers=['VCU', 'VDC'], comment='è¡¨å¾\x81é©¾é©¶å\x91\x98å\x88¶å\x8a¨è¯·æ±\x82ç\x9a\x84æ\x8e¨æ\x9d\x86è¡\x8cç¨\x8b\nSignal that corrVCUonds to the driver brake request', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={4095: 'Init/Error'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('-5.000000'), min=Decimal('-5'), max=Decimal('47')), Signal(name='BotBotBrkPedlTrvlOutp', start_bit=20, size=12, is_little_endian=False, is_signed=False, offset=Decimal('-5'), factor=Decimal('0.015625'), unit='mm', receivers=['VCU', 'VDC'], comment='è¡¨å¾\x81å®\x9eé\x99\x85æ\x8e¨å\x8a¨æ¶²å\x8e\x8bæµ\x81é\x87\x8fç\x9a\x84æ\x8e¨æ\x9d\x86è¡\x8cç¨\x8b\nSignal that represents the actual shifted hydraulic volume', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={4095: 'Init/Error'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('-5.000000'), min=Decimal('-5'), max=Decimal('47')), Signal(name='BotBotBrkPedlTrvlOutpVldy', start_bit=38, size=2, is_little_endian=False, is_signed=False, offset=Decimal('0'), factor=Decimal('1'), unit='', receivers=['VCU', 'VDC'], comment='EbbBrkPedlTrvlOutpä¿¡å\x8f·æ\x9c\x89æ\x95\x88æ\x80§\nQualifier for sigEbbBrkPedlTrvlOutp', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={0: 'Not initialized', 1: 'Normal', 2: 'Faulty', 3: 'Reserved'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('0'), min=Decimal('0'), max=Decimal('3')), Signal(name='BotBotPedlTrvlOutpDrvrVldy', start_bit=36, size=2, is_little_endian=False, is_signed=False, offset=Decimal('0'), factor=Decimal('1'), unit='', receivers=['VCU', 'VDC'], comment='EbbBrkPedlTrvlOutpDrvrä¿¡å\x8f·æ\x9c\x89æ\x95\x88æ\x80§\nQualifier for sigEbbBrkPedlTrvlOutpDrvr', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={0: 'Not initialized', 1: 'Normal', 2: 'Faulty', 3: 'Reserved'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('0'), min=Decimal('0'), max=Decimal('3')), Signal(name='BotBotExtReqSts', start_bit=46, size=2, is_little_endian=False, is_signed=False, offset=Decimal('0'), factor=Decimal('1'), unit='', receivers=['VCU', 'VDC'], comment='eBoosterå¤\x96é\x83¨å\x88¶å\x8a¨è¯·æ±\x82å\x8a\x9fè\x83½ç\x8a¶æ\x80\x81\nSignal to indicate that an external brake request can be executed by the eBooster', multiplex=None, mux_value=None, is_float=False, is_ascii=False, type_label='', enumeration=None, comments={}, attributes={'GenSigStartValue': '0', 'GenSigSendType': 'Cyclic'}, values={0: 'Not initialized', 1: 'Not available', 2: 'Available', 3: 'Reserved'}, mux_val_grp=[], muxer_for_signal=None, calc_min_for_none=True, calc_max_for_none=True, cycle_time=0, initial_value=Decimal('0'),

ebroecker, 2025-01-02:
Hi @CXZ1988 ,

the motorola startbit is arbitrariness of candb++.
canmatrix writes the correct startbit to dbc but uses an other (for our point of view more logical) internal representation. 
Exporting to dbc will write the startbit as dbc expects it. 

I just discussed it here:  #821

to get the same startbit as candb++ shows you (with default settings) just use `get_startbit(True,True)` method of the signal object.

there is also information about byte-order and starbits here: 
https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder

CXZ1988, 2025-01-03:
Hello @ebroecker , 
Thanks very much for your feedback. It is clear from my side now. Have a nice day.🤝