ebroecker/canmatrix #833: Wrong start bit for auth info signal

Patryk-97, 2025-01-16:
Since auth info signal is encoded in Motorola format then start bit should indicate to most significant bit. But in the code start bit points to least significant bit

xRowe, 2025-01-23:
Thanks for your feedback

I validate on CANoe, it should be OK now
Please Try [xRowe:Fix_#833](https://github.com/ebroecker/canmatrix/pull/835/commits/f67419a64bb57755019512fcbe627ee80188ba95)

Patryk-97, 2025-01-27:
Hello @xRowe. But still it's not what I meant. Freshness and auth info signals are encoded in Motorola byte order format. So start bit of those signals indicates to most significant bit. So start bit of freshness signal should be:
start_bit = int(ipdu_length, 0)*8 + 7 (assuming it exists)

And start bit of auth info signal should be:
start_bit = int(ipdu_length, 0)*8 + int(freshness_tx_length, 0) + 7

[image]
Reference to AUTOSAR system template: https://www.autosar.org/fileadmin/standards/R22-11/CP/AUTOSAR_TPS_SystemTemplate.pdf

xRowe, 2025-02-11:
For the start-bit of Motorola format in canmatrix, 
I think you can refer to 
#829
#821

Hope it is the answer.

ebroecker, 2026-11-16:
Hi @Patryk-97,

thanks for the explanation and sorry for the late reply.

I checked this again with some examples, and I think both numbers describe the same bit positions, only in different numbering conventions.

canmatrix stores the startbit of motorola signals internally as position of the first (MSB-first transmitted) bit inside the frame byte stream, and not as LSB-0 position of the MSB (this is the dbc/autosar convention you used in your calculation).

so with your example (8 byte payload, 8 bit freshness, 24 bit auth info):

Freshness: internal startbit = payload*8 (=64), exported to dbc/arxml you get 71 = payload*8 + 7
AuthInfo: internal startbit = payload*8 + freshness_len (=72), exported you get 79 = payload*8 + freshness_len + 7
so the exported startbits are exactly the values from your calculation. also @xRowe validated the layout against CANoe.

you can convert the internal representation yourself with:

```python
signal.get_startbit()                  # internal: stream position of first transmitted bit
signal.get_startbit(bit_numbering=1)   # dbc/autosar: LSB-0 position of the MSB
signal.get_startbit(bit_numbering=1, start_little=True) # LSB-0 position of the LSB
```

so I think this works as intended. if you still see a real mismatch on the wire, please share the arxml and the CANoe trace, then I will happily reopen

closed 2026-11-16 by ebroecker