ebroecker/canmatrix #837: Option to hide "xxx is not supported"

hongbo-miao, 2025-01-27:
Would it be a good idea to hide these info by default or provide an option? Thanks! ☺️

```sh
ldf is not supported
xls is not supported
xlsx is not supported
```

tuna-f1sh, 2025-01-30:
The module is using `logging` https://github.com/ebroecker/canmatrix/blob/fc2cf58a18c40e827438f334dc458d9edff75ed0/src/canmatrix/formats/__init__.py#L27 . The default level is `WARNING` so this prints.

One control the logging and remove this in their code before importing canmatrix with:

`logging.getLogger("canmatrix").setLevel(logging.CRITICAL)`

hongbo-miao, 2025-02-03:
Thank you so much @tuna-f1sh !

I am wondering if it is a good idea to hide these warnings by default? Only when read `ldf`, `xls`, `xlsx`, throw error and remind people to install extra dependencies? Thanks!

tuna-f1sh, 2025-02-03:
The commit  https://github.com/ebroecker/canmatrix/commit/d9902ace093fc1bb31af02599243847b529b0b44 means that they will be hidden by default since the default logging level is < DEBUG. It hasn't been included in a release yet however since the last was over one year ago - not sure why?

morbult, 2025-02-04:
@ebroecker, would it be possible to make a new release in the near future?

That way users will not be puzzled anymore by these warnings every time canmatrix is imported somewhere. Unfortunatley, this has been an issue for a long time, #684, [old reddit post](https://www.reddit.com/r/learnpython/comments/l2b6py/asammdf_error_saying_xls_is_not_supported_with/). It is especially confusing when you are not even importing `canmatrix` directly, but some other package that depends on it, for example `asammdf`. No indication where the messages come from or explanation why those formats would be needed.

ebroecker, 2025-02-10:
new release 1.1 is out

closed 2025-02-10 by ebroecker