ebroecker/canmatrix #847: Convert kcd to DBC: No "GenMsgCycleTime" records

MamoruAsagami, 2025-03-05:
When you convert from .kcd to .dbc, message cycle time info is lost.
Message cycle times are coded as Message.interval as in  <Message ... interval="100"> in .kcd, which should be transformed as below in .dbc. 

BA_DEF_ BO_  "GenMsgCycleTime" INT 0 0;
BA_DEF_DEF_  "GenMsgCycleTime" 0;
BA_ "GenMsgCycleTime" BO_ <can id 1> 100;
BA_ "GenMsgCycleTime" BO_ <can id 2> 50;
...

ebroecker, 2025-03-18:
[example.zip](https://github.com/user-attachments/files/19327681/example.zip)

Hi @MamoruAsagami
I can't reproduce this issue. The attached example seems to convert correct to:
```
...
BO_ 16 SomeMessageName: 8 Vector__XXX
 SG_ someSignalName : 24|32@1- (1,0) [-2147483648|2147483647] "W" Vector__XXX

...

BA_DEF_ BO_ "GenMsgCycleTime" INT 0 65535;
BA_ "GenMsgCycleTime" BO_ 16 100;
```

MamoruAsagami, 2025-03-19:
I should have said 'Convert kcd to DBC: No "GenMsgCycleTime" records when Message elements don't have triggered attribute.'  
See the attached zip file.
issue_847.kcd ... What I received.
issue_847.dbc ... I used canmatrix to convert issue_847.kcd.  No problems.
issue_847a.kcd ... I used CANBabel to convert issue_847.dbc.  No problems.
issue_847a.dbc ... I used canmatrix to convert issue_847a.kcd.  No "GenMsgCycleTime" records!
issue_847LessTriggered.kcd ... I edited issue_847.kcd to delete triggered="true" of <Message ... >.
issue_847LessTriggered.dbc ... I used canmatrix to convert issue_847LessTriggered.kcd.  The problem was reproduced.
Incidentally, when I changed to triggered="false" the conversion showed no problems.

[issue847ToReproduce.zip](https://github.com/user-attachments/files/19331214/issue847ToReproduce.zip)

ebroecker, 2025-03-19:
Hi @MamoruAsagami 

thanks for clearification, so we don't need to have the "triggered" attribute to know it's a cyclic message.
I'll provide a pr next

ebroecker, 2025-03-19:
Hopefully fixed in branch:

https://github.com/ebroecker/canmatrix/tree/issue_847

MamoruAsagami, 2025-04-25:
I confirmed the problem was solved in the branch issue_847.

Since I'm not sure if I am eligible to  "Close with comment", I simply "Comment."